orangepi-xunlong/orangepi-build #158: 编译kernel_5.10.160 for opi5plus无法正常启动

TangToutou, 2024-05-14:
按照用户手册编译过6.1内核和5.10内核，其中6.1正常使用，但是5.10内核编译出linux-image-legacy-rockchip-rk3588_1.0.8_arm64.deb并替换后，无法正常开机使用，会卡在：
[    4.771136] pwr       0x014c: 0000e839 00000011 00000000 00000000
[    4.771670] status    0x0180: 0011e839 00000000 00000000 00000000
[    4.772204] req       0x010c: 00001400 00000000 00000000 00021c0f
[    4.772738] idle      0x0120: 00021c0f 00000000 0000ffff 0000ffff
[    4.773272] ack       0x0118: 00021c0f 00000000 00021c0f 00000000
[    4.773805] mem_pwr   0x01a0: 00000000 00000000 00000000 00000000
[    4.774339] chain_st  0x01f0: ee17c6ff 000003ff 11e83900 00000000
[    4.774872] mem_st    0x01f8: 11e83900 00000000 00000203 0000ffff
[    4.775405] repair_st 0x0290: fff70be1 000301ff 00000000 00000000
[    4.775939] THERMAL REGS:
[    4.776181] 00000000: 00000100 00000001 0000000c 0000007f 00000000 00000001 00000000 0000007f
[    4.776927] 00000020: 00000000 0000007e 00000000 00000129 00000129 00000129 00000129 00000129
[    4.777673] 00000040: 00000129 00000129 00000000 00000000 00000000 00000000 00000000 00000000
[    4.778419] 00000060: 00000000 00000000 00000000 00000153 00000000 00000000 00000000 00000000
[    4.779162] 00000080: 00000000 00000000
[    4.779502] Kernel Offset: disabled
[    4.779809] CPU features: 0x000,48000026,6a00aa18
[    4.780228] Memory Limit: none
[    4.783345] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b ]---

此外，官方镜像：6.1内核无法正常调用NPU2和NPU3

baiywt, 2025-04-08:
@TangToutou 我测了5.10内核没遇到这个问题，你dtb有同步更新吗？

TangToutou, 2025-04-08:
> [@TangToutou](https://github.com/TangToutou) 我测了5.10内核没遇到这个问题，你dtb有同步更新吗？

不好意思，没有及时更新上述问题，该问题在官方镜像的1.2.0版本源码中已经修复