orangepi-xunlong/orangepi-build #160: Feature Request: Enable Ceph + CephFS + RBD Drivers / Kernel Modules

nikos445, 2024-05-19:
Enable Ceph + CephFS + RBD + NFSv4 -> Drivers / Kernel Modules by default for OrangePI CM4 
Tested it and it works perfectly!

nikos445, 2025-04-26:
Sorry, I searched a lot, unfortunately I didn't keep a list of changes that I did when building the image.

I used the ./build.sh and build the full OS image (debian) and options I selected by navigating the menus in terminal.
I remember enabling some kernel settings (or setting it as modules) for: CephFS, RBD, NFSv4. 

I can provide more details from inside my custom image / kernel, if you need something specific. Its live, up and running.

masterlog80, 2026-03-04:
Hello @nikos445,
May I ask you what is the kernel version?
And if NPU works?