orangepi-xunlong/orangepi-build #161: Orange pi 5 plus problem with the install built linux-headers-current-rockchip-rk3588_1.0.8_arm64.deb

ZombieXLX, 2024-05-23:
I build u-boot:
~/orangepi-build$ sudo ./build.sh  BOARD=orangepi5plus BRANCH=current BUILD_OPT=u-boot
i get:
~/orangepi-build$ file u-boot/v2017.09-rk3588/scripts/kconfig/conf.o
u-boot/v2017.09-rk3588/scripts/kconfig/conf.o: ELF 64-bit LSB relocatable, x86-64, version 1 (SYSV), not stripped
I can install my build u-boot:
~/orangepi-build$ sudo apt purge -y linux-u-boot-orangepi5plus-current
~/orangepi-build$ sudo dpkg -i linux-u-boot-current-orangepi5plus_1.0.8_arm64.deb
~/orangepi-build$ sudo nand-sata-install
and that work it's fine.

I build kernel:
sudo ./build.sh  BOARD=orangepi5plus BRANCH=current BUILD_OPT=kernel KERNEL_CONFIGURE=no
i get:
~/orangepi-build$ file kernel/orange-pi-6.1-rk35xx/scripts/kconfig/conf.o
kernel/orange-pi-6.1-rk35xx/scripts/kconfig/conf.o: ELF 64-bit LSB relocatable, x86-64, version 1 (SYSV), not stripped
it's build work's fine too:
~/orangepi-build$ sudo apt purge -y linux-image-current-rockchip-rk3588
~/orangepi-build$ sudo dpkg -i linux-image-current-rockchip-rk3588_1.0.8_arm64.deb

BUT when i try install headers...
~/orangepi-build$ sudo dpkg -i linux-headers-current-rockchip-rk3588_1.0.8_arm64.deb
I get:
Compiling headers - please wait ...
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
make[1]: *** [scripts/Makefile.host:120: scripts/kconfig/conf] Error 1
make: *** [Makefile:704: oldconfig] Error 2
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
make[2]: *** [scripts/Makefile.host:120: scripts/kconfig/conf] Error 1
make[1]: *** [Makefile:704: syncconfig] Error 2
make: *** [Makefile:805: include/config/auto.conf.cmd] Error 2
make: *** [include/config/auto.conf.cmd] Deleting file «include/generated/rustc_cfg»
make: *** [include/config/auto.conf.cmd] Deleting file «include/generated/autoconf.h»

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make: *** [Makefile:813: include/config/auto.conf] Error 1

dali0302, 2024-07-17:
Sorry to bother you. I have the same problem, have you solved it?

ZombieXLX, 2024-07-17:
> Sorry to bother you. I have the same problem, have you solved it?

Nope

dali0302, 2024-07-18:
There are two versions of the orangepi ubuntu os, v5.10.160 and v6.1.43. I met the issue in the v6.1.43, and I reinstall the kernel5.10.43 ubuntu22.04 image. The problem disappeared.
So i decided to use the kernel5.10.43 image.

baiywt, 2025-02-17:
The issue has been resolved in the latest 6.1 kernel code. Please update the kernel to the latest version and recompile the headers.Thanks.

--- a/scripts/package/builddeb
+++ b/scripts/package/builddeb
@@ -64,6 +64,7 @@ EOT
 cat >> $pdir/DEBIAN/postinst << EOT
 cd /usr/src/linux-headers-$version
 rm -rf scripts/kconfig/conf scripts/basic/fixdep scripts/mod/mk_elfconfig
+find -name *.o |xargs rm -rf
 echo "Compiling headers - please wait ..."
 find -type f -exec touch {} +
 yes "" | make oldconfig >/dev/null

closed 2025-02-17 by ZombieXLX