orangepi-xunlong/orangepi-build #162: libGLES_mali.so with OpenCL for Orangepi Zero 3

firefish37, 2024-05-24:
Hello, I need library "libGLES_mali.so" with OpenCL support for Orangepi Zero 3 Android, where can I get one? 

Version that is in android image has no openCL, no functions cl*** shown by " nm -D  libGLES_mali.so" it has only gl.. 

Ive spent about two weeks searching for one that will fit, but not found any

Tryed to use some from https://developer.arm.com/downloads/-/mali-drivers/user-space, but got ERROR: The DDK (built for 0x60000000 r0p0 status range [0..15]) is not compatible with this Mali GPU device, /dev/mali0 detected as 0x7093 r0p0

It seems they built for another hardware...
Is where a way to run OpenCL on Orangepi Zero 3, Mali G31 mp2 Android?

loki666, 2024-11-27:
@firefish37 did you find the lib?

firefish37, 2024-11-29:
> @firefish37 did you find the lib?

No