orangepi-xunlong/orangepi-build #165: OrangePi CM4  build img fail: [ error ] Installation of /root/orangepi-jammy-desktop-xfce_1.0.2_all.deb failed [ orangepicm4 jammy yes rockchip-rk356x ]

Aliaszhang, 2024-06-25:
Error message:

Displaying message: Building desktop package orangepi-jammy-desktop-xfce_1.0.2_all info
Displaying message: Creating board support package for desktop  info
Displaying message: Starting rootfs and image building process for legacy orangepicm4 jammy null xfce no info
Displaying message: Extracting jammy-xfce-arm64.239...3f4.tar.lz4 126 days old info
Displaying message: Applying distribution specific tweaks for jammy info
Displaying message: Applying common tweaks  info
Displaying message: Cleaning package lists
Displaying message: Updating package lists
Displaying message: Temporarily disabling initramfs-tools hook for kernel
Displaying message: Installing PACKAGE_LIST_FAMILY packages ethtool can-utils
Displaying message: Removing PACKAGE_LIST_FAMILY_REMOVE packages mpv
Displaying message: Installing linux-u-boot-legacy-orangepicm4_1.0.2_arm64.deb
Displaying message: Installing linux-image-legacy-rockchip-rk356x_1.0.2_arm64.deb
Displaying message: Installing linux-dtb-legacy-rockchip-rk356x_1.0.2_arm64.deb
Displaying message: Installing orangepi-bsp-cli-orangepicm4_1.0.2_arm64.deb
Displaying message: Installing orangepi-jammy-desktop-xfce_1.0.2_all.deb
Displaying message: ERROR in function install_deb_chroot build.sh:305 -> scripts/main.sh:631 -> scripts/main.sh:600 -> scripts/debootstrap.sh:64 -> scripts/distributions.sh:347 -> scripts/image-helpers.sh:221 -> scripts/general.sh:0 err
Displaying message: **Installation of /root/orangepi-jammy-desktop-xfce_1.0.2_all.deb failed** orangepicm4 jammy yes rockchip-rk356x err
Displaying message: Process terminated  info
Displaying message: unmount_on_exit() called! build.sh:305 -> scripts/main.sh:1 -> scripts/image-helpers.sh:0 err
Displaying message: Unmounting /home/abc/abc_workspace/orangepi/source/orangepi-build/.tmp/rootfs-107a4c47-32fe-4d2b-ac42-592d86c4e33f/ info
Displaying message: ERROR in function unmount_on_exit build.sh:305 -> scripts/main.sh:1 -> scripts/image-helpers.sh:90 -> scripts/general.sh:0 err
Displaying message: debootstrap-ng was interrupted  err
Displaying message: Process terminated  info

I referenced those links , none of one works for me.
[](https://github.com/orangepi-xunlong/orangepi-build/issues/98)
[](https://github.com/orangepi-xunlong/orangepi-build/issues/135)
[](http://forum.orangepi.cn/thread/3102)

AliasPineapple, 2024-07-08:
Anyone can help?  I've updated to the newest version.
```
[ o.k. ] Applying common tweaks 
[ .... ] Cleaning [ package lists ]
[ .... ] Updating [ package lists ]
[ .... ] Temporarily disabling [ initramfs-tools hook for kernel ]
[ .... ] Installing PACKAGE_LIST_FAMILY packages [ ethtool can-utils ]
[ .... ] Removing PACKAGE_LIST_FAMILY_REMOVE packages [ mpv ]
[ .... ] Installing [ linux-u-boot-legacy-orangepicm4_1.0.6_arm64.deb ]
[ .... ] Installing [ linux-image-legacy-rockchip-rk356x_1.0.6_arm64.deb ]
[ .... ] Installing [ linux-dtb-legacy-rockchip-rk356x_1.0.6_arm64.deb ]
[ .... ] Installing [ orangepi-bsp-cli-orangepicm4_1.0.6_arm64.deb ]
[ .... ] Installing [ orangepi-jammy-desktop-xfce_1.0.6_all.deb ]
[ error ] ERROR in function install_deb_chroot [ build.sh:305 -> scripts/main.sh:639 -> scripts/main.sh:608 -> scripts/debootstrap.sh:64 -> scripts/distributions.sh:347 -> scripts/image-helpers.sh:221 -> scripts/general.sh:0 ]
[ error ] Installation of /root/orangepi-jammy-desktop-xfce_1.0.6_all.deb failed [ orangepicm4 jammy yes rockchip-rk356x ]
[ o.k. ] Process terminated 
[ error ] unmount_on_exit() called! [ build.sh:305 -> scripts/main.sh:1 -> scripts/image-helpers.sh:0 ]
[ o.k. ] Unmounting [ /home/abc/abc_workspace/orangepi/source/orangepi-build/.tmp/rootfs-fd952f78-5ff2-4a37-a447-843c7d2d0f6d/ ]
[ error ] ERROR in function unmount_on_exit [ build.sh:305 -> scripts/main.sh:1 -> scripts/image-helpers.sh:90 -> scripts/general.sh:0 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated 
```

AliasPineapple, 2024-07-22:
I rollback to a previous version and the result is compilation pass. Then, I go back to the current version, It also compily pass!!!

WTF

closed 2024-07-22 by AliasPineapple