orangepi-xunlong/orangepi-build #167: device-mapper missing from orange pi 5 plus 32gb

samrocketman, 2024-07-05:
# OS

Ubuntu

# Package

```
# apt-cache show linux-image-current-rockchip-rk3588
Package: linux-image-current-rockchip-rk3588
Status: install ok installed
Priority: optional
Section: kernel
Installed-Size: 113653
Maintainer: Orange Pi <leeboby@aliyun.com>
Architecture: arm64
Source: linux-6.1.43-rockchip-rk3588
Version: 1.0.8
Description: Linux kernel, version 6.1.43-rockchip-rk3588
 This package contains the Linux kernel, modules and corresponding other
 files, version: 6.1.43-rockchip-rk3588.
Description-md5: 556d1c3248b99a15012a77d57ccce977
Homepage: https://www.kernel.org/
```

# Error

```
# cryptsetup luksOpen /dev/nvme0n1p2 encrypted
Cannot initialize device-mapper. Is dm_mod kernel module loaded?
Cannot use device encrypted, name is invalid or still in use.
Enter passphrase for /dev/nvme0n1p2:
```

Attempt to load module:

```
# modprobe dm_mod
modprobe: FATAL: Module dm_mod not found in directory /lib/modules/6.1.43-rockchip-rk3588
```

samrocketman, 2024-07-05:
I notice there's `/opt/linux-headers-current-rockchip-rk3588_1.0.8_arm64.deb` on my orangepi but I don't think there's a source package so I can't recompile the kernel directly on the board easily.

felixw7k, 2024-09-28:
Same on the opi5-max 8gb.
bump for a kernel with device-mapper-support ;)

samrocketman, 2024-09-29:
I was able to work around by compiling

Devric, 2024-12-28:
> I was able to work around by compiling

Hi Can you write a simple guide how to do this?
is it simply just build the main branch of this repo?

samrocketman, 2024-12-28:
@Devric asked:

> Hi Can you write a simple guide how to do this?
> is it simply just build the main branch of this repo?

Sure, here are my notes on that subject.  I plan to publish a few blog posts on it but you are welcome to read my draft notes.  https://github.com/samrocketman/blog/blob/main/_drafts/orangepi-5-plus-32gb.md#add-encypted-drives-by-compiling-linux-kernel

Devric, 2025-01-01:
> @Devric asked:
>
>
> Sure, here are my notes on that subject. I plan to publish a few blog posts on it but you are welcome to read my draft notes. [...]

Perfect, am able to compile with nfs4.2 and now able to run kubernetes and longhorn

samrocketman, 2025-01-02:
Amazing, glad it helped you.