orangepi-xunlong/orangepi-build #169: Unable to build orange pi linux for orange pi zero 2w

legut2, 2024-07-11:
I have tried building orange pi linux for zero 2w board. I haven't been able to do it with either this repo or the following: https://github.com/orangepi-xunlong/linux-orangepi/tree/orange-pi-6.1-sun50iw9

I can't even get a verification email for forums to orangepi.org.

I am attempting to get buildroot to have orangepi zero 2w to work with it. It doesn't seem to have the drivers on the def config.

Any help is appreciated.

I'll get errors like this:
```
2024-07-06T15:07:16.2466302Z + [ arm64 = um ]
2024-07-06T15:07:16.2466675Z + wl=
2024-07-06T15:07:16.2468387Z + ld=/home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/host/bin/aarch64-buildroot-linux-gnu-ld
2024-07-06T15:07:16.2471589Z + ldflags=-EL  -maarch64elf -z noexecstack --no-warn-rwx-segments --no-undefined -X -shared -Bsymbolic -z notext  --no-apply-dynamic-relocs --fix-cortex-a53-843419 --build-id=sha1 --orphan-handling=warn
2024-07-06T15:07:16.2473743Z + ldlibs=
2024-07-06T15:07:16.2475912Z + ldflags=-EL  -maarch64elf -z noexecstack --no-warn-rwx-segments --no-undefined -X -shared -Bsymbolic -z notext  --no-apply-dynamic-relocs --fix-cortex-a53-843419 --build-id=sha1 --orphan-handling=warn --script=./arch/arm64/kernel/vmlinux.lds
2024-07-06T15:07:16.2478008Z + [ .tmp_vmlinux.kallsyms1 != 1 ]
2024-07-06T15:07:16.2480417Z + ldflags=-EL  -maarch64elf -z noexecstack --no-warn-rwx-segments --no-undefined -X -shared -Bsymbolic -z notext  --no-apply-dynamic-relocs --fix-cortex-a53-843419 --build-id=sha1 --orphan-handling=warn --script=./arch/arm64/kernel/vmlinux.lds --strip-debug
2024-07-06T15:07:16.2482565Z + is_enabled CONFIG_VMLINUX_MAP
2024-07-06T15:07:16.2483323Z + grep -q ^CONFIG_VMLINUX_MAP=y include/config/auto.conf
2024-07-06T15:07:16.2488950Z + /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/host/bin/aarch64-buildroot-linux-gnu-ld -EL -maarch64elf -z noexecstack --no-warn-rwx-segments --no-undefined -X -shared -Bsymbolic -z notext --no-apply-dynamic-relocs --fix-cortex-a53-843419 --build-id=sha1 --orphan-handling=warn --script=./arch/arm64/kernel/vmlinux.lds --strip-debug -o .tmp_vmlinux.kallsyms1 --whole-archive vmlinux.a .vmlinux.export.o init/version-timestamp.o --no-whole-archive --start-group ./drivers/firmware/efi/libstub/lib.a --end-group
2024-07-06T15:07:16.9260057Z /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/host/bin/aarch64-buildroot-linux-gnu-ld: Unexpected GOT/PLT entries detected!
2024-07-06T15:07:16.9265631Z /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/host/bin/aarch64-buildroot-linux-gnu-ld: Unexpected run-time procedure linkages detected!
2024-07-06T15:07:16.9269440Z /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/host/bin/aarch64-buildroot-linux-gnu-ld: ID map text too big or misaligned
2024-07-06T15:07:17.1645527Z /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/host/bin/aarch64-buildroot-linux-gnu-ld: drivers/net/phy/sunxi-ephy.o: in function `ephy_config_init':
2024-07-06T15:07:17.1649926Z /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/build/linux-custom/drivers/net/phy/sunxi-ephy.c:163:(.text+0x5f4): undefined reference to `sun50i_ephy_calibrate_value'
2024-07-06T15:07:17.1654148Z /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/host/bin/aarch64-buildroot-linux-gnu-ld: drivers/net/phy/sunxi-ephy.o: in function `ac200_ephy_enable':
2024-07-06T15:07:17.1658743Z /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/build/linux-custom/drivers/net/phy/sunxi-ephy.c:354:(.text+0x900): undefined reference to `ac200_enable'
2024-07-06T15:07:17.1664467Z /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/host/bin/aarch64-buildroot-linux-gnu-ld: /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/build/linux-custom/drivers/net/phy/sunxi-ephy.c:357:(.text+0x928): undefined reference to `ac200_enable'
2024-07-06T15:07:17.1671988Z /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/host/bin/aarch64-buildroot-linux-gnu-ld: /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/build/linux-custom/drivers/net/phy/sunxi-ephy.c:382:(.text+0x9f0): undefined reference to `sun50i_ephy_calibrate_value'
2024-07-06T15:07:17.5438801Z make[2]: *** [scripts/Makefile.vmlinux:34: vmlinux] Error 1
2024-07-06T15:07:17.5441061Z make[1]: *** [Makefile:1255: vmlinux] Error 2
2024-07-06T15:07:17.5441789Z make[1]: *** Waiting for unfinished jobs....
2024-07-06T15:07:17.9968652Z make[1]: Leaving directory '/home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/build/linux-custom'
2024-07-06T15:07:17.9972188Z make: *** [package/pkg-generic.mk:283: /home/runner/work/buildroot-orange-pi-zero-2w-dial-gauge-reader/buildroot-orange-pi-zero-2w-dial-gauge-reader/output/build/linux-custom/.stamp_built] Error 2
```

Which might mean that the repo has failed to include all the files when building orange-pi-6.1-sun50iw9.

defencedog, 2024-07-29:
@legut2 I am facing a [similar error](https://github.com/orangepi-xunlong/orangepi-build/issues/173)

legut2, 2024-07-29:
@defencedog I eventually figured out how to build but it wasn't intuitive and it was a bit of guess and checking for what to select within the menus prior to the build. It seems inconsistent. I will get back to focusing on build system once I finish other work, because I'd like to get the specifics of this build working on buildroot because then I get benefits of community contributions of buildroot.

leeboby, 2025-02-17:
It is recommended to use the following kernel configuration when compiling the kernel source separately.

https://github.com/orangepi-xunlong/orangepi-build/blob/next/external/config/kernel/linux-6.1-sun50iw9-next.config