orangepi-xunlong/orangepi-build #175: Bug while building Gnome Jammy Image

defencedog, 2024-08-03:
```
Unpacking libjbig2dec0:arm64 (0.19-3build2) ...
Selecting previously unselected package libgs9:arm64.
Preparing to unpack .../0212-libgs9_9.55.0~dfsg1-0ubuntu5.9_arm64.deb ...
Unpacking libgs9:arm64 (9.55.0~dfsg1-0ubuntu5.9) ...
/home/orangepi/Downloads/orangepi-build/scripts/debootstrap.sh: line 324: 246367 Killed                  LC_ALL=C LANG=C chroot $SDCARD /bin/bash -e -c "DEBIAN_FRONTEND=noninteractive apt-get -y -q                           $apt_extra $apt_
extra_progress install ${apt_desktop_install_flags} $PACKAGE_LIST_DESKTOP"
                                                                          [ o.k. ] Uninstall packages [ gdm3 ]
                                                                                                              E: Could not get lock /var/lib/dpkg/lock. It is held by process 247848 (dpkg)
                                                                   E: Unable to lock the administration directory (/var/
lib/dpkg/), is another process using it?
                                        [ error ] ERROR in function create_rootfs_cache [ ./build.sh:305 -> scripts/main.sh:624 -> scripts/main.sh:593 -> scripts/debootstrap.sh:50 -> scripts/debootstrap.sh:342 -> scripts/general.sh:0 ]
                                                                                                                   [ error ] Installation of Orange Pi packages failed
                                               [ o.k. ] Process terminated
                                                                           [ error ] unmount_on_exit() called! [ ./build.sh:305 -> scripts/main.sh:1 -> scripts/image-helpers.sh:0 ]
                                                            [ o.k. ] Unmounting [ /home/orangepi/Downloads/orangepi-build/.tmp/rootfs-5aaaa659-5a7d-4ef6-816d-cc3a75db8ddd/ ]
                                                     [ error ] ERROR in function unmount_on_exit [ ./build.sh:305 -> scripts/main.sh:1 -> scripts/image-helpers.sh:90 -> scripts/general.sh:0 ]
                                                                       [ error ] debootstrap-ng was interrupted
                                                                                                                [ o.k. ]
 Process terminated
```

baiywt, 2025-04-08:
@defencedog It’s recommended to build the image on a PC running Ubuntu 22.04 for a smoother experience and better performance. If the issue still occurs, feel free to share the details.

defencedog, 2025-04-11:
Resolved with latest images

closed 2025-04-11 by defencedog