orangepi-xunlong/orangepi-build #183: Orangepi5plus_1.0.10_ubuntu_jammy_desktop_xfce_linux6.1.43 如何解决linux-headers的问题？

qw4wer, 2024-09-09:
**板子是orange pi 5 plus 16g的。使用镜像为Orangepi5plus_1.0.10_ubuntu_jammy_desktop_xfce_linux6.1.43**

想使用redroid，Android运行不起来，根据文档需要安装扩展
```
apt install linux-modules-extra-`uname -r`
modprobe binder_linux devices="binder,hwbinder,vndbinder"
modprobe ashmem_linux
```
运行无法安装
```
root@orangepi5plus:/data/anbox-modules# apt install linux-modules-extra-`uname -r`
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-modules-extra-6.1.43-rockchip-rk3588
E: Couldn't find any package by glob 'linux-modules-extra-6.1.43-rockchip-rk3588'
```
运行下列命令没有返回
```
lsmod | egrep "ashmem|binder"
```
现在通过想安装anbox-modules解决，运行./INSTALL.sh提示需要安装内核头文件
```
Error! Your kernel headers for kernel 6.1.43-rockchip-rk3588 cannot be found.
Please install the linux-headers-6.1.43-rockchip-rk3588 package or use the --kernelsourcedir option to tell DKMS where it's located.
Error! Your kernel headers for kernel 6.1.43-rockchip-rk3588 cannot be found.
Please install the linux-headers-6.1.43-rockchip-rk3588 package or use the --kernelsourcedir option to tell DKMS where it's located.
modprobe: FATAL: Module ashmem_linux not found in directory /lib/modules/6.1.43-rockchip-rk3588
modprobe: FATAL: Module binder_linux not found in directory /lib/modules/6.1.43-rockchip-rk3588
ls: cannot access '/dev/binder': No such file or directory
ls: cannot access '/dev/ashmem': No such file or directory
```

 但是安装自带的/opt/linux-headers-current-rockchip-rk3588_1.0.10_arm64.deb安装报错。
```
root@orangepi5plus:/opt# apt install ./linux-headers-current-rockchip-rk3588_1.0.10_arm64.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'linux-headers-current-rockchip-rk3588' instead of './linux-headers-current-rockchip-rk3588_1.0.10_arm64.deb'
The following NEW packages will be installed:
  linux-headers-current-rockchip-rk3588
0 upgraded, 1 newly installed, 0 to remove and 169 not upgraded.
Need to get 0 B/12.7 MB of archives.
After this operation, 84.6 MB of additional disk space will be used.
Get:1 /opt/linux-headers-current-rockchip-rk3588_1.0.10_arm64.deb linux-headers-current-rockchip-rk3588 arm64 1.0.10 [12.7 MB]
Selecting previously unselected package linux-headers-current-rockchip-rk3588.
(Reading database ... 162682 files and directories currently installed.)
Preparing to unpack .../linux-headers-current-rockchip-rk3588_1.0.10_arm64.deb ...
Unpacking linux-headers-current-rockchip-rk3588 (1.0.10) ...
Setting up linux-headers-current-rockchip-rk3588 (1.0.10) ...
Compiling headers - please wait ...
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
make[1]: *** [scripts/Makefile.host:120: scripts/kconfig/conf] Error 1
make: *** [Makefile:704: oldconfig] Error 2
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: scripts/kconfig/conf.o: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
make[2]: *** [scripts/Makefile.host:120: scripts/kconfig/conf] Error 1
make[1]: *** [Makefile:704: syncconfig] Error 2
make: *** [Makefile:805: include/config/auto.conf.cmd] Error 2
make: *** [include/config/auto.conf.cmd] Deleting file 'include/generated/rustc_cfg'
make: *** [include/config/auto.conf.cmd] Deleting file 'include/generated/autoconf.h'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make: *** [Makefile:813: include/config/auto.conf] Error 1

```
安装了上面的头文件，运行anbox-modules的安装脚本报错如下:
```
root@orangepi5plus:/data/anbox-modules# ./INSTALL.sh

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=6.1.43-rockchip-rk3588 all KERNEL_SRC=/lib/modules/6.1.43-rockchip-rk3588/build...(bad exit status: 2)
ERROR (dkms apport): binary package for anbox-ashmem: 1 not found
Error! Bad return status for module build on kernel: 6.1.43-rockchip-rk3588 (aarch64)
Consult /var/lib/dkms/anbox-ashmem/1/build/make.log for more information.

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=6.1.43-rockchip-rk3588 all KERNEL_SRC=/lib/modules/6.1.43-rockchip-rk3588/build...(bad exit status: 2)
ERROR (dkms apport): binary package for anbox-binder: 1 not found
Error! Bad return status for module build on kernel: 6.1.43-rockchip-rk3588 (aarch64)
Consult /var/lib/dkms/anbox-binder/1/build/make.log for more information.
modprobe: FATAL: Module ashmem_linux not found in directory /lib/modules/6.1.43-rockchip-rk3588
modprobe: FATAL: Module binder_linux not found in directory /lib/modules/6.1.43-rockchip-rk3588
ls: cannot access '/dev/binder': No such file or directory
ls: cannot access '/dev/ashmem': No such file or directory
```

运行命令查询内核及头文件版本
```
root@orangepi5plus:/data/anbox-modules# dpkg --list | grep 'Linux kernel'
ii  cpufrequtils                           008-2build1                                 arm64        utilities to deal with the cpufreq Linux kernel feature
ii  kmod                                   29-1ubuntu1                                 arm64        tools for managing Linux kernel modules
ii  libcpufreq0                            008-2build1                                 arm64        shared library to deal with the cpufreq Linux kernel feature
ii  libsctp1:arm64                         1.0.19+dfsg-1build1                         arm64        user-space access to Linux kernel SCTP - shared library
ii  linux-headers-current-rockchip-rk3588  1.0.10                                      arm64        Linux kernel headers for 6.1.43-rockchip-rk3588 on arm64
ii  linux-image-current-rockchip-rk3588    1.0.10                                      arm64        Linux kernel, version 6.1.43-rockchip-rk3588
```
**请问如何解决上面的问题？**

ps:我尝试自己编译内核，使用orangepi-build编译，但是编译出来的版本是1.0.8的。
```
[ o.k. ] Building deb [ linux-u-boot-current-orangepi5plus_1.0.8_arm64.deb ]
[ o.k. ] U-boot build done [ @host ]
[ o.k. ] Target directory [ /data/orangepi-build/output/debs/u-boot ]
[ o.k. ] File name [ linux-u-boot-current-orangepi5plus_1.0.8_arm64.deb ]
[ o.k. ] Runtime [ 0 min ]
[ o.k. ] Repeat Build Options [ sudo ./build.sh  BOARD=orangepi5plus BRANCH=current BUILD_OPT=u-boot  ]
```
本地仓库版本
```
root@ubuntu22:/data/orangepi-build/kernel/orange-pi-6.1-rk35xx# git branch -vv
* （头指针在 FETCH_HEAD 分离） b3424fb44 Update for opicm5

```
**请问是哪里配置有问题吗？能麻烦提供下解决方案吗？**

holai, 2024-12-30:
有解决办法了吗  我也遇到了

baiywt, 2025-02-17:
这个问题最新的镜像已经解决了。

closed 2025-03-17 by orangepi-xunlong