orangepi-xunlong/orangepi-build #185: Orange Pi 5 Pro Camera work bad

vzhikh, 2024-09-12:
Hello,
Clear image install: 
`Orangepi5pro_1.0.4_ubuntu_jammy_desktop_xfce_linux6.1.43.img`

Enabled cam1 in orangepi-config.

Test command: 
`gst-launch-1.0 v4l2src device=/dev/video11 io-mode=4 ! video/x-raw,format=NV12,width=2112,height=1568,framerate=30/1 ! mpph264enc ! rtph264pay ! udpsink host=10.10.2.222 port=3000`

It works, but only 15 fps and rkaiq doesn't work.

[image]

jackwei86, 2024-12-02:
hi,bro.i met the same problem. The ov13850 or ov13855  only 15fps. Have you solved this problem?

vzhikh, 2024-12-02:
Hello, no, same 15 fps on the last image(

jackwei86, 2024-12-04:
> Hello, no, same 15 fps on the last image(

OK.thx

mike00340, 2024-12-23:
I'm having the same difficulties.
The picture is too dark, only very bright objects are visible.

fedya, 2025-01-31:
hi guys
looks like i found a culrpit of the problem
can you provide here your:

- uname -a
- dtb file you used for a board
- and type of your board orangepi5 or orangepi5b
- your distro

fedya, 2025-02-18:
@lucle-bonboncar 
rebuild your kernel with 
`# CONFIG_RT_GROUP_SCHED is not set
`
if you can't do it try to add cgroup_disable=cpu to your cmdline

lucle-bonboncar, 2025-02-18:
> [@lucle-bonboncar](https://github.com/lucle-bonboncar) rebuild your kernel with `# CONFIG_RT_GROUP_SCHED is not set ` if you can't do it try to add cgroup_disable=cpu to your cmdline

hello, do you have any instruction for me can follow, i am not familier with the linux kernel

fedya, 2025-02-18:
@lucle-bonboncar you use armbian or what?

AlvinJ15, 2025-11-16:
Hi @fedya  I'm facing same issue with the OV13850 camera:
**uname -a:**
Linux orangepi5pro 6.1.43-rockchip-rk3588 #1.0.4 SMP Fri Jul  5 01:17:22 CST 2024 aarch64 aarch64 aarch64 GNU/Linux
**Board:**
Opi 5 Pro V1.2
**OS:**
Orangepi5pro_1.0.4_ubuntu_jammy_desktop_xfce_linux6.1.43.img

This is my current orangepiEnv.txt after added cgroup_disable=cpu (AFTER REBOOT):
```
cat /boot/orangepiEnv.txt 
verbosity=1
bootlogo=true
extraargs=cma=128M cgroup_disable=cpu
overlay_prefix=rk3588
fdtfile=rockchip/rk3588s-orangepi-5-pro.dtb
rootdev=UUID=99a4ebdd-350c-4623-aa2c-d152a279947a
rootfstype=ext4
overlays=opi5pro-cam1
```
Command:
```
gst-launch-1.0 -v v4l2src device=/dev/video11 io-mode=4 ! video/x-
```
OUTPUT:
```
raw,format=NV12,width=2112,height=1568,framerate=30/1 ! fpsdisplaysink text-overlay=false video-sink=fakesink
Setting pipeline to PAUSED ...
Using mplane plugin for capture 
Pipeline is live and does not need PREROLL ...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFakeSink:fakesink0: sync = true
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = video/x-raw, format=(string)NV12, width=(int)2112, height=(int)1568, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)1:3:5:1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw, format=(string)NV12, width=(int)2112, height=(int)1568, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)1:3:5:1
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink.GstProxyPad:proxypad0: caps = video/x-raw, format=(string)NV12, width=(int)2112, height=(int)1568, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)1:3:5:1
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFakeSink:fakesink0.GstPad:sink: caps = video/x-raw, format=(string)NV12, width=(int)2112, height=(int)1568, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)1:3:5:1
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink: caps = video/x-raw, format=(string)NV12, width=(int)2112, height=(int)1568, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)1:3:5:1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-raw, format=(string)NV12, width=(int)2112, height=(int)1568, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)1:3:5:1
Redistribute latency...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFakeSink:fakesink0: sync = true
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 9, dropped: 0, current: 16.87, average: 16.87
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 17, dropped: 0, current: 15.02, average: 15.95
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 25, dropped: 0, current: 15.02, average: 15.64
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 33, dropped: 0, current: 15.02, average: 15.49
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 41, dropped: 0, current: 15.02, average: 15.39
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 49, dropped: 0, current: 15.02, average: 15.33
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 57, dropped: 0, current: 15.02, average: 15.29
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 65, dropped: 0, current: 15.02, average: 15.26
```

I want to test the Promised: 13mp@30fps,ISP OFF on OV13850 camera with a orangepi5 board
I tried to disable in different ways the ISP but it always output the same 15FPS, even tried to reduce resolution to see if the FPS is increased but can't archieve this.

Even I tried to built the image from this repository but it produces a 8gb .img and when trying to burn it on the SD card the orangepi doesn't start, So I stuck with the option of using a pre-built image.