orangepi-xunlong/orangepi-build #194: 系统启动时，有大量无效的系统文件修改，会导致意外断电后，系统损坏无法启动

kingsimba, 2024-10-24:
有些启动脚本中，会用 sed 修改一些系统文件。虽然其实一个字节也没有变，但是文件还是被重写了。我们遇到了很多突然断电的情况下，导致这些文件全都变成乱码。轻则一些 log 文件无限变大，占满整个磁盘，重则使得系统无法启动。

https://github.com/orangepi-xunlong/orangepi-build/blob/f11df9bf0d0bd57f6c968b4c520c2ced0517baeb/external/packages/bsp/common/usr/lib/orangepi/orangepi-hardware-optimization#L54

希望改成：

1.  先在内存中生成修改后的结果，把结果和现有文件比较。如果有变化再写入。
2. 写入时，先生成 tmp 临时文件，再 rename。从而最大程度上避免文件变成乱码。

希望能指导我们一下，系统中还有哪些类似的地方，我们希望先全部禁用掉。

laishere, 2025-02-09:
因为写入磁盘是内核负责的，断电的情况下内核也无法正常工作了。
所以如果断电了，那正在写入的文件就肯定存在损坏的风险，除非有备用电源继续维持工作直到正常关机。

不过先保存到tmp我感觉是有一定的可行性的，但是也是无法避免文件损坏的问题。问题在于，临时文件损坏了，还可否被正常清理。

另外，你可以先检查一下当前使用的文件系统是否支持日志，不支持的话你可以尝试使用支持日志的文件系统并启用日志功能。

baiywt, 2025-04-08:
@kingsimba 一般orangepiEnv.txt文件损坏会导致系统无法启动，后面我们把orangepi-hardware-optimization中的add_usb_storage_quirks注释了，你现在还会有orangepiEnv.txt损坏的问题吗？