orangepi-xunlong/orangepi-build #195: please update the rknpu version to 0.98

kiron111, 2024-10-26:
if the rknpu version is old, all the new rkllm model will not worked
pls update it, thankyou

the rknpu 0.9.8 driver is here
https://github.com/airockchip/rknn-llm/tree/main/rknpu-driver

and pls refer to the offical doc
https://github.com/airockchip/rknn-llm/tree/main/doc
quote from the offiical doc:
```
补充说明： 若是发现编译内核的过程中，出现如图 2-1 的错误日志： 图 2-1 内核编译错误 用户需要修改内核头文件 kernel/include/linux/mm.h，加入以下代码：
static inline void vm_flags_set(struct vm_area_struct *vma, vm_flags_t flags)
{ vma->vm_flags |= flags; }
static inline void vm_flags_clear(struct vm_area_struct *vma, vm_flags_t flags)
{ vma->vm_flags &= ~flags; }
```

fuzhenxiao, 2024-10-27:
I agree, currently the rknpu driver is 0.9.8, while the driver here is not updated yet.

LuckyMouseLai, 2024-10-30:
i agree,  0.9.8 is needed.

aileen5150, 2024-12-26:
编译rknpu 0.9.6 可以正常通过
rknpu 0.9.8 遇到 如下错误，不知道如何解决
[ error ] ERROR in function compile_kernel [ ./build.sh:305 -> scripts/main.sh:638 -> scripts/main.sh:541 -> scripts/compilation.sh:506 -> scripts/general.sh:0 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated

aileen5150, 2024-12-26:
问题解决了，参考 https://blog.csdn.net/weixin_74253011/article/details/141058714
rknpu_devfreq.c:
static const struct rockchip_opp_data rk3576_npu_opp_data = {
    .set_read_margin = rk3576_npu_set_read_margin,
//     .set_soc_info = rockchip_opp_set_low_length, 未定义, 需要删除
#if KERNEL_VERSION(6, 1, 0) <= LINUX_VERSION_CODE
    .config_regulators = npu_opp_config_regulators,
    .config_clks = npu_opp_config_clks,
#endif
};

更新后查看板子上的/sys/kernel/debug/rknpu/version
显示 RKNPU driver: v0.9.8

orangepi-xunlong, 2025-03-17:
已更新。

closed 2025-03-17 by orangepi-xunlong

hajajmaor, 2025-03-29:
Can you please add the fix and release another OS version?