orangepi-xunlong/orangepi-build #198: [BUG]: `/dev/loop0` do not exist

annadostoevskaya, 2024-11-03:
During the image file build process, the build.sh script encountered an error, as `loop0` do not exist.

baiywt, 2025-04-08:
@annadostoevskaya Thanks for the report. It seems the Docker environment you're using doesn't provide /dev/loop0, which is needed for image creation. Try running Docker with --privileged and /dev mounted. We’ll improve support for Docker in future updates.

annadostoevskaya, 2025-04-08:
> @annadostoevskaya Thanks for the report. It seems the Docker environment you're using doesn't provide /dev/loop0, which is needed for image creation. Try running Docker with --privileged and /dev mounted. We’ll improve support for Docker in future updates.

What do you think about this: 
https://github.com/orangepi-xunlong/orangepi-build/pull/197
?