orangepi-xunlong/orangepi-build #199: Build uboot for CM4 error

trananhdung0991, 2024-11-07:
I have build linux image for orangepi CM4 and facing this issue. Can you help me resolve this issue?
Output:
make[2]: 'arch/arm/dts/rk3566-orangepi-3b.dtb' is up to date.
start=$(ccache aarch64-none-linux-gnu-nm u-boot | grep __rel_dyn_start | cut -f 1 -d ' '); end=$(ccache aarch64-none-linux-gnu-nm u-boot | grep __rel_dyn_end | cut -f 1 -d ' '); tools/relocate-rela u-boot-nodtb.bin 0x00a00000 $start $end
  COPY    u-boot.dtb
  MKIMAGE u-boot.itb
  CC      spl/common/spl/spl.o
  CC      spl/arch/arm/cpu/armv8/fwcall.o
  LD      spl/arch/arm/cpu/armv8/built-in.o
  CC      spl/lib/display_options.o
  LD      spl/lib/built-in.o
  LD      spl/common/spl/built-in.o
  LD      spl/u-boot-spl
  OBJCOPY spl/u-boot-spl-nodtb.bin
  CAT     spl/u-boot-spl-dtb.bin
  COPY    spl/u-boot-spl.bin
[ error ] ERROR in function compile_uboot [ ./build.sh:305 -> scripts/main.sh:631 -> scripts/main.sh:517 -> scripts/compilation.sh:247 -> scripts/general.sh:0 ]
[ error ] U-boot compilation failed 
[ o.k. ] Process terminated 

cat output/debug/compilation.log
tools/rockchip/boot_merger.c: In function ‘main’:
tools/rockchip/boot_merger.c:895:18: warning: array subscript 20 is outside array bounds of ‘char[20]’ [-Warray-bounds]
  895 |         str[len] = 0;
      |         ~~~~~~~~~^~~
tools/rockchip/boot_merger.c:933:14: note: while referencing ‘name’
  933 |         char name[MAX_NAME_LEN];
      |              ^~~~
FATAL ERROR: Couldn't open "./14:26": No such file or directory
./tools/mkimage: Can't read u-boot.itb.tmp: Invalid argument
make: *** [Makefile:1053: u-boot.itb] Error 255
make: *** Waiting for unfinished jobs....

baiywt, 2025-02-17:
Hello, built on which system?

trananhdung0991, 2025-02-17:
build on Ubuntu 22.04

baiywt, 2025-04-08:
@trananhdung0991 I tried compiling it, and it worked fine on my side.
[image]