orangepi-xunlong/orangepi-build #204: build debian img for Orange pi 3B (OS Debian 12 LXDE)

zeljkoc, 2024-12-03:
Help my:

Step by step:

Full OS image for flashing
  Do not change the kernel configuration 
orangepi3b         Rockchip  RK3566 quad core 2-8GB RAM GBE eMMC USB3 NvMe WiFi/BT 
legacy  Old stable / Legacy 
 bookworm Debian 12 Bookworm 
 no  Image with console interface (server)
yes Minimal image with console interface

./build.sh

[ warn ] This script requires root privileges, trying to use sudo 
[ o.k. ] Using config file [ ...../orangepi-build/userpatches/config-example.conf ]
[ .... ] Extension being added [ rkbin-tools :: added by build.sh:305 -> scripts/main.sh:374 -> scripts/configuration.sh:151 -> external/config/sources/families/rockchip-rk356x.conf:1 -> external/config/sources/families/include/rockchip64_common.inc:5 -> scripts/extensions.sh:0 ]
[ o.k. ] Extension manager [ processed 3 Extension Methods calls and 3 Extension Method implementations ]
[ o.k. ] Preparing [ host ]
[ o.k. ] Build host OS release [ bookworm ]
[ error ] ERROR in function prepare_host [ build.sh:305 -> scripts/main.sh:638 -> scripts/main.sh:426 -> scripts/general.sh:1468 -> scripts/general.sh:0 ]
[ error ] It seems you ignore documentation and run an unsupported build system: bookworm 
[ o.k. ] Process terminated

zeljkoc, 2024-12-03:
same problem on debian 11 lxqt

baiywt, 2025-02-17:
The build host is incorrect; it should be a PC with Ubuntu 22.04 (Jammy) installed.

closed 2025-03-17 by orangepi-xunlong