orangepi-xunlong/orangepi-build #205: orangepi5 kernel compiled fail

sunnythree, 2024-12-10:
system: ubuntu 20.04
compiled with default config.
./build.sh
->Kernel package
->Do not change the kernel configuration
->orangepi5          Rockchip  RK3588S octa core 4-16GB RAM GBE USB3 USB-C NVMe
->current Recommended. Come with best support

error log:

drivers/gpu/arm/bifrost/backend/gpu/mali_kbase_pm_metrics.c:115:9: internal compiler error: Illegal instruction
make[5]: *** [scripts/Makefile.build:250: drivers/gpu/arm/bifrost/backend/gpu/mali_kbase_pm_metrics.o] Error 1
make[4]: *** [scripts/Makefile.build:500: drivers/gpu/arm/bifrost] Error 2
make[3]: *** [scripts/Makefile.build:500: drivers/gpu/arm] Error 2
make[2]: *** [scripts/Makefile.build:500: drivers/gpu] Error 2
drivers/net/wireless/intel/iwlwifi/iwl-eeprom-parse.c:129:9: internal compiler error: Illegal instruction
make[6]: *** [scripts/Makefile.build:250: drivers/net/wireless/intel/iwlwifi/iwl-eeprom-parse.o] Error 1
make[5]: *** [scripts/Makefile.build:500: drivers/net/wireless/intel/iwlwifi] Error 2
make[4]: *** [scripts/Makefile.build:500: drivers/net/wireless/intel] Error 2
kernel/sched/core.c:10448:17: internal compiler error: Illegal instruction
make[3]: *** [scripts/Makefile.build:250: kernel/sched/core.o] Error 1
make[2]: *** [scripts/Makefile.build:500: kernel/sched] Error 2
make[1]: *** [scripts/Makefile.build:500: kernel] Error 2
make[3]: *** [scripts/Makefile.build:500: drivers/net/wireless] Error 2
make[2]: *** [scripts/Makefile.build:500: drivers/net] Error 2
make[1]: *** [scripts/Makefile.build:500: drivers] Error 2

by the way:
I do not compile sucessfully anyway after I have try lots times. can anyone help me?

leeboby, 2025-02-17:
orangepi-build only supports running on an Ubuntu 22.04 host.

ektosMV, 2025-02-20:
I have the same problem with the 22.04. 6.8.0-52-generic #53~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC x86_64 x86_64 x86_64 GNU/Linux
Branch: orange-pi-6.1-sun50iw9

  AR      drivers/dma-buf/built-in.a
make[1]: *** [scripts/Makefile.build:500: drivers] Error 2

ektosMV, 2025-02-20:
I've found the solution: the reason is AMD CPU.
I followed this instruction:
https://forums.gentoo.org/viewtopic-p-8772505.html

baiywt, 2025-04-08:
Build works fine on my side.!

[image]