orangepi-xunlong/orangepi-build #208: 编译ubuntu桌面系统报错

wenhuiwang93, 2025-01-22:
我该从哪里找到libmali-valhall-g610-g13p0-x11-gbm_1.9-1_arm64.deb这个文件？或者说我该在哪里添加一些配置来生成这个文件？

[error msg]:
Displaying message: Installing orangepi-firmware_1.0.8_all.deb
Displaying message: Installing orangepi-config_1.0.8_all.deb
Displaying message: Installing orangepi-zsh_1.0.8_all.deb
Displaying message: Installing orangepi-plymouth-theme_1.0.8_all.deb
Displaying message: Installing wiringpi_2.57.deb
Displaying message: Installing libmali-valhall-g610-g13p0-x11-gbm_1.9-1_arm64.deb
Displaying message: ERROR in function dpkg_install_deb_chroot ./build.sh:305 -> scripts/main.sh:638 -> scripts/main.sh:607 -> scripts/debootstrap.sh:64 -> scripts/distributions.sh:464 -> external/config/sources/families/include/rockchip64_common.inc:277 -> external/config/sources/families/include/rk35xx_gpu_vpu.inc:62 -> scripts/image-helpers.sh:243 -> scripts/general.sh:0 err
Displaying message: Installation of /root/libmali-valhall-g610-g13p0-x11-gbm_1.9-1_arm64.deb failed orangepi5b jammy yes rockchip-rk3588 err
Displaying message: Process terminated  info
Displaying message: unmount_on_exit() called! ./build.sh:305 -> scripts/main.sh:1 -> scripts/image-helpers.sh:0 err
Displaying message: Unmounting /home/ax650n/Work/rk3588s/orangepi-build/.tmp/rootfs-a81e16bb-ae7d-46d7-85fd-4286ec08e084/ info
Displaying message: ERROR in function unmount_on_exit ./build.sh:305 -> scripts/main.sh:1 -> scripts/image-helpers.sh:90 -> scripts/general.sh:0 err
Displaying message: debootstrap-ng was interrupted  err
Displaying message: Process terminated  info

wenhuiwang93, 2025-01-23:
solved.  missed files are here: https://github.com/orangepi-xunlong/rk-rootfs-build

closed 2025-01-23 by wenhuiwang93

LY226-max, 2025-02-11:
你好，我和你遇到了同样的问题，当前我的确是找到了这个包libmali-valhall-g610-g13p0-x11-gbm_1.9-1_arm64.deb。编译镜像前我看了报错的路径，手动创建并存放这个包后，重新编译的过程中，创建的路径和包又丢失了，请问你是怎么解决的？这是脚本的问题么？

LY226-max, 2025-02-11:
Hello, I've encountered the same problem as you. Currently, I've indeed found the package `libmali-valhall-g610-g13p0-x11-gbm_1.9-1_arm64.deb`. Before compiling the image, I checked the path mentioned in the error message, manually created the path and placed the package there. However, during the re - compilation process, the created path and the package disappeared again. Have you encountered this problem? How did you handle it? Is it an issue with the script?

wenhuiwang93, 2025-02-11:
> Hello, I've encountered the same problem as you. Currently, I've indeed found the package `libmali-valhall-g610-g13p0-x11-gbm_1.9-1_arm64.deb`. Before compiling the image, I checked the path [...]

I think network updates overwrote your actions. 
If you are manually placing the relevant files, you might as well try setting the IGNORE_UPDATES variable in userpatches/config-default.conf to "yes".

LY226-max, 2025-02-12:
Thank you so much for your suggestions. They have successfully solved my problem.