orangepi-xunlong/orangepi-build #211: orangepi5pro image build fail

RomanSereda, 2025-02-07:
./build.sh -> 
--> Full OS image for flashing
--> orangepi5pro
--> current Recommended. Come with best support
--> Ubuntu jammy 22.04 LTS
--> no  Image with console interface (server) 
-->no  Standard image with console interface

Tail log:

  AR      drivers/thermal/built-in.a
/bin/sh: 1: Syntax error: Unterminated quoted string
make[6]: *** [drivers/gpu/arm/mali400/mali/Kbuild:254: drivers/gpu/arm/mali400/mali/__malidrv_build_info.c] Error 2
make[5]: *** [scripts/Makefile.build:500: drivers/gpu/arm/mali400/mali] Error 2
make[4]: *** [scripts/Makefile.build:500: drivers/gpu/arm/mali400] Error 2
make[3]: *** [scripts/Makefile.build:500: drivers/gpu/arm] Error 2
make[2]: *** [scripts/Makefile.build:500: drivers/gpu] Error 2
make[1]: *** [scripts/Makefile.build:500: drivers] Error 2
make: *** [Makefile:2012: .] Error 2
[ error ] ERROR in function compile_kernel [ ./build.sh:305 -> scripts/main.sh:638 -> scripts/main.sh:541 -> scripts/compilation.sh:506 -> scripts/general.sh:0 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated

leeboby, 2025-02-17:
Is it running on Ubuntu 22.04 Host?

RomanSereda, 2025-02-18:
> Is it running on Ubuntu 22.04 Host?

yes

baiywt, 2025-04-08:
@RomanSereda 
To resolve this issue, please ensure that your orangepi-build is up to date. You can do this by running the following commands:
```
cd orangepi-build
git pull origin
```
After that, delete the kernel directory:
`rm -rf kernel/orange-pi-6.1-rk35xx`

Then, try recompiling again. This should help resolve the error.

closed 2025-04-11 by RomanSereda