orangepi-xunlong/orangepi-build #221: orangepi4a problem with MPI7002

michaelslobodskih, 2025-03-22:
Good day, the board does not work with a 7" LCD screen (1024x600). When turned on, the message "Not supported" appears on the screen. I tried with different screens MPI7002 and MPI7006, with the orangepi3b board these screens work fine.

baiywt, 2025-04-08:
@michaelslobodskih Hi, this may be a compatibility issue — 1024x600 resolution is currently not well supported on this board.

michaelslobodskih, 2025-04-08:
@baiywt And in fact I know how to partially solve this problem, but I want to get the right solution to the problem from the developers.

Here solution for kernel
https://github.com/walnutpi/linux/commit/64bf5787418ba2609119fded095efa580a816a12

But same solution does not work for Uboot (if you add this line to the corresponding hdmi uboot driver)

michaelslobodskih, 2025-04-08:
@baiywt Screens with this resolution work fine on orangepi3b (both on uboot and kernel level). I want it to work the same way on their new orangepi4a board :)

defencedog, 2025-04-08:
@michaelslobodskih I too like the overall design of 4A ... however have you been able to get VPU working either in mpv or ffplay or chrome?

michaelslobodskih, 2025-04-10:
> [@michaelslobodskih](https://github.com/michaelslobodskih) Hi, this may be a compatibility issue — 1024x600 resolution is currently not well supported on this board.

I see you have come a long way in orangepi4a, tell me if there is any progress on my problem? These screens are very common now and many people connect them - they are disappointed that the screen does not work with the board. This is very demotivating. Above I described how to solve this issue for the kernel, but this "trick" does not solve the problem with display in uboot (let's imagine that I am planning to launch my device for sale and I want to show my logo to my users in a human way while the board is loading, and loading uboot takes a very significant time, while now the user gets a "black screen" and does not understand whether the device has turned on or not.

baiywt, 2025-04-10:
@michaelslobodskih I am currently in the process of acquiring a screen for testing, and it should take a few days to arrive. I’ll keep you updated on the progress once I’ve had the chance to test it.

michaelslobodskih, 2025-04-24:
@baiywt hello , is there any good news ? Also I still don't quite understand why orangepi4a takes so much time to render the first image? even uboot. In this regard, orangepi 3b renders the boot logo much faster. I saw in the build settings there is a file mentioning fastboot, but how to use it and how to use it is not clear.

defencedog, 2025-04-24:
@michaelslobodskih my 4a boots properly. I am not able to boot from SPI however I am using NVME+sdcard combo

michaelslobodskih, 2025-05-19:
@baiywt hello, is there any news ?

defencedog, 2025-05-21:
@michaelslobodskih boots from NVME + sdcard ...but not from SPI