orangepi-xunlong/orangepi-build #223: orangepi5pro image boot fail

ARC-MX, 2025-03-31:
system: ubuntu 20.04
compiled with default config.

sudo ./build.sh BOARD=orangepi5pro BRANCH=current BUILD_OPT=image RELEASE=bookworm BUILD_MINIMAL=no BUILD_DESKTOP=yes KERNEL_CONFIGURE=no DESKTOP_ENVIRONMENT=xfce DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base DESKTOP_APPGROUPS_SELECTED="internet"

->Kernel package
->Do not change the kernel configuration
->orangepi5 Rockchip RK3588S octa core 4-16GB RAM GBE USB3 USB-C NVMe
->current Recommended. Come with best support

I compiled the orangepi5pro image successfully, but after downloading it, it couldn't start. The serial port showed the following message:

Serial port start log：

```
SoftReset, 18603923 us
DDR 9fffbe1e78 cym 24/02/04-10:09:20,fwver: v1.16
LPDDR5, 2400MHz
channel[0] BW=16 Col=10 Bk=16 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=4096MB
channel[1] BW=16 Col=10 Bk=16 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=4096MB
channel[2] BW=16 Col=10 Bk=16 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=4096MB
channel[3] BW=16 Col=10 Bk=16 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=4096MB
Manufacturer ID:0xff
CH0 RX Vref:25.8%, TX Vref:22.0%,22.0%
CH1 RX Vref:27.9%, TX Vref:19.0%,19.0%
CH2 RX Vref:25.8%, TX Vref:20.0%,19.0%
CH3 RX Vref:28.4%, TX Vref:22.0%,22.0%
change to F1: 534MHz
change to F2: 1320MHz
change to F3: 1968MHz
change to F0: 2400MHz
out
U-Boot SPL board init
U-Boot SPL 2017.09-orangepi (Mar 31 2025 - 22:03:00)
unrecognized JEDEC id bytes: ff, ff, ff
Trying to boot from MMC1
spl: mmc init failed with error: -123
Trying to boot from MMC2
Card did not respond to voltage select!
spl: mmc init failed with error: -95
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###
# Reset the board to bootrom #
```
After that, it was stuck.
I want to know how to solve this problem.

baiywt, 2025-04-08:
@ARC-MX Is the system installed on eMMC or an SD card (TF card)? Based on the log, it seems like neither the eMMC nor the SD card is being recognized properly. If you are using the official pre-built image from the website, does this issue still occur?

williamso1982, 2025-05-04:
I got the same problem after the recently update from git source. what happen?

broadblues, 2025-09-01:
by comparison the same section of the successful boot of the prebuilt image 

```
DDR 9fffbe1e78 cym 24/02/04-10:09:20,fwver: v1.16
LPDDR5, 2400MHz
channel[0] BW=16 Col=10 Bk=16 CS0 Row=15 CS=1 Die BW=16 Size=1024MB
channel[1] BW=16 Col=10 Bk=16 CS0 Row=15 CS=1 Die BW=16 Size=1024MB
channel[2] BW=16 Col=10 Bk=16 CS0 Row=15 CS=1 Die BW=16 Size=1024MB
channel[3] BW=16 Col=10 Bk=16 CS0 Row=15 CS=1 Die BW=16 Size=1024MB
Manufacturer ID:0xff
CH0 RX Vref:27.1%, TX Vref:21.0%,0.0%
CH1 RX Vref:28.5%, TX Vref:20.0%,0.0%
CH2 RX Vref:27.9%, TX Vref:20.0%,0.0%
CH3 RX Vref:27.9%, TX Vref:21.0%,0.0%
change to F1: 534MHz
change to F2: 1320MHz
change to F3: 1968MHz
change to F0: 2400MHz
out
U-Boot SPL board init
U-Boot SPL 2017.09-orangepi (Jul 09 2024 - 16:38:40)
unknown raw ID 0 0 0
unrecognized JEDEC id bytes: 00, 00, 00
Trying to boot from MMC1
Trying fit image at 0x4000 sector
## Verified-boot: 0
## Checking atf-1 0x00040000 ... sha256(6dbe36e3e6...) + OK
## Checking uboot 0x00200000 ... sha256(3289d770ea...) + OK
## Checking fdt 0x00349678 ... sha256(1b12611ccd...) + OK
## Checking atf-2 0xff100000 ... sha256(40e9180d15...) + OK
## Checking atf-3 0x000f0000 ... sha256(d189664c33...) + OK
Jumping to U-Boot(0x00200000) via ARM Trusted Firmware(0x00040000)
Total: 218.881 ms
'''

UBOOT output to full boot follows this.

FriendlyNGeeks, 2025-11-03:
just to confirm everyone trying to build from ubuntu 20.04 jammy is failing to cimpile? i'm not alone? ive also spun up a docker jammy and still failing