orangepi-xunlong/orangepi-build #227: After using new boot from nvme feature OPI4a is no longer booting without a display attached

qr0n, 2025-04-12:
the issue is as stated in the title, its not booting. all i see from the lights is the red one on constantly, the blue one blinking twice then nothing. i know its not the power supply since it works when theres a display attached. it doesnt connect to a network, i dont have a serial to usb board thingy ma bober. sorry i couldnt provide that information.

qr0n, 2025-04-18:
update: i figured out i can use another device to get serial out, here is the data
```
Welcome to minicom 2.8

OPTIONS: I18n
Port /dev/serial0, 03:19:20

Press CTRL-A Z for help on special keys

àø[152]HELLO! BOOT0 is starting!
[155]BOOT0 commit : 33d8db429a
[159]periph0 has been enabled
[162]set pll end
[164]PL gpio voltage : 3.3V
[168][pmu]: bus read error
[171]PMU: AXP2202
[176]PMU: AXP1530
[181]power mode:33, sys_vol:920
[186]vaild para:1  select dram para0
[189]board init ok
[212]enable_jtag
[214]boot param - This media is not supported
[219]boot param - Media not supportedsunxi_bootparam_cpu_read...91
[225]boot param - magic error
[229]DRAM BOOT DRIVE INFO: V0.67
[233]DRAM_VCC set to 1160 mv
[360]DRAM retraining ten
[377][AUTO DEBUG]32bit,2 ranks training success!
[407]Soft Training Version: T2.0
[7288][SOFT TRAINING] CLK=1200M Stable memtest pass
[7293]DRAM CLK =1200 MHZ
[7296]DRAM Type =8 (3:DDR3,4:DDR4,7:LPDDR3,8:LPDDR4)
[7306]DRAM SIZE =4096 MBytes, para1 = 310a, para2 = 10001000, tpr13 = 6461
[7334]DRAM simple test OK.
[7337]dram size =4096
[7344]nsi init 2023-2-23
[7347]set spi freq:100000000
[7351]spi sample_mode:1 sample_delay:1f
[7356]spinor id is: 20 41 18, read cmd: 0b
[7360]Succeed in reading toc file head.
[7364]The size of toc is 1d4000.
[7675]Entry_name        = u-boot
[7685]Entry_name        = monitor
[7688]Entry_name        = scp
[7692]Entry_name        = dtb
[7696]Jump to second Boot.
NOTICE:  BL31: v2.5(debug):20a8ac62a
NOTICE:  BL31: Built : 17:17:28, Jan 22 2024
NOTICE:  BL31: No DTB found.
nsi init ok 2022-11-08
ERROR:   Error initializing runtime service opteed_fast

U-Boot 2018.07-orangepi-config-dirty (Apr 08 2025 - 22:49:32 +0800) Allwinner Technology

[07.843]CPU:   Allwinner Family
[07.846]Model: sun55iw3
I2C:   ready
[07.856]DRAM:  4 GiB
[07.861]Relocation Offset is: 75e2f000
[07.921]secure enable bit: 0
[07.925]PMU: AXP2202
[07.931]BMU: AXP2202
[07.935][AXP2202] comm status : 0x0 = 0x20, 0x1 = 0x90
[07.940][AXP2202] onoff status: 0x20 = 0x4, 0x21 = 0x0
[07.945][AXP2202] reboot/charge status: 0xf0 = 0x0
[07.949]PMU: AXP1530
[07.952]CPU=1008 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=600Mhz
[07.958]gic: sec monitor mode
SPI ALL:   ready
[07.999]flash init start
[... 424 lines cut ...]
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    24877568 Bytes = 23.7 MiB
   Load Address: 41000000
   Entry Point:  41000000
   Verifying Checksum ... Bad Data CRC
ERROR: can't get kernel image!
SCRIPT FAILED: continuing...
starting USB...
USB0:   start sunxi  USB-DRD...
config usb clk ok
sunxi USB-DRD init ok...
USB EHCI 1.00
scanning bus 0 for devices... 1 USB Device(s) found
USB1:   start sunxi  USB1-Host...
config usb clk ok
sunxi USB1-Host init ok...
USB EHCI 1.00
scanning bus 1 for devices... 2 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found

Device 0: device type unknown
... is now current device
** Bad device usb 0 **
** Bad device usb 0 **
=>```

defencedog, 2025-04-18:
The last of logs is similar to https://github.com/orangepi-xunlong/orangepi-build/issues/226#issuecomment-2797993609

Indicating that `/boot` partition cannot be found. Try inserting sdcard & boot from it while `/` resides on nvme

P.S I am not able to use SPI flash as `/boot`

qr0n, 2025-04-18:
got it! trying it out now (before i settled for using nvme as /home)