orangepi-xunlong/orangepi-build #232: Help!!!!!!!orangepi5pro image boot fail

williamso1982, 2025-05-04:
I compiled the orangepi5pro image successfully, but after downloading it, it couldn't start. The serial port showed the following message:
BTW, the old image file that I complied from Jun 2024 is no problem. what happen for the recently update...................
Serial port start log：
DDR d5483af87d cym 23/11/23-16:15:24,fwver: v1.15
LPDDR5, 2736MHz
channel[0] BW=16 Col=10 Bk=16 CS0 Row=16 CS=1 Die BW=16 Size=2048MB
channel[1] BW=16 Col=10 Bk=16 CS0 Row=16 CS=1 Die BW=16 Size=2048MB
channel[2] BW=16 Col=10 Bk=16 CS0 Row=16 CS=1 Die BW=16 Size=2048MB
channel[3] BW=16 Col=10 Bk=16 CS0 Row=16 CS=1 Die BW=16 Size=2048MB
Manufacturer ID:0xff
CH0 RX Vref:25.4%, TX Vref:21.0%,0.0%
CH1 RX Vref:27.1%, TX Vref:21.0%,0.0%
CH2 RX Vref:23.8%, TX Vref:18.0%,0.0%
CH3 RX Vref:25.8%, TX Vref:21.0%,0.0%
change to F1: 528MHz
change to F2: 1320MHz
change to F3: 1968MHz
change to F0: 2736MHz
out
Boot1 Release Time: Nov 22 2022 14:37:46, version: 1.10
Emmc IO init.
Emmc IO init.
mmc_set_bus_width: 1
SetEmmcClk: 375000, 2, 64
SetEmmcClk: 375000, 2, 64
mmc_set_bus_width: 8
SetEmmcClk: 200000000, 0, 3
Enable PHY CLK: 200000000, timing: 9
SdmmcInit=2 0
BootCapSize=100000
UserCapSize=59000MB
FwPartOffset=2000 , 100000
UsbBoot ...53917
powerOn 56280
Emmc IO init.
SoftReset, 7545781 us
DDR 9fffbe1e78 cym 24/02/04-10:09:20,fwver: v1.16
LPDDR5, 2400MHz
channel[0] BW=16 Col=10 Bk=16 CS0 Row=16 CS=1 Die BW=16 Size=2048MB
channel[1] BW=16 Col=10 Bk=16 CS0 Row=16 CS=1 Die BW=16 Size=2048MB
channel[2] BW=16 Col=10 Bk=16 CS0 Row=16 CS=1 Die BW=16 Size=2048MB
channel[3] BW=16 Col=10 Bk=16 CS0 Row=16 CS=1 Die BW=16 Size=2048MB
Manufacturer ID:0xff
CH0 RX Vref:26.7%, TX Vref:20.0%,0.0%
CH1 RX Vref:27.9%, TX Vref:21.0%,0.0%
CH2 RX Vref:25.4%, TX Vref:18.0%,0.0%
CH3 RX Vref:27.5%, TX Vref:20.0%,0.0%
change to F1: 534MHz
change to F2: 1320MHz
change to F3: 1968MHz
change to F0: 2400MHz
out
U-Boot SPL board init
U-Boot SPL 2017.09-orangepi (May 04 2025 - 22:17:41)
unrecognized JEDEC id bytes: ff, ff, ff
Trying to boot from MMC1
spl: mmc init failed with error: -123
Trying to boot from MMC2
Card did not respond to voltage select!
spl: mmc init failed with error: -95
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###
# Reset the board to bootrom #

broadblues, 2025-05-13:
I'm also having this type of issue.

I can boot my Orange Pi 5 pro from the generic images provided by orangepi, but when I build an image via thi build system, it does not boot. Noting the OP comment about June 2024 above I checked out the version of orangepi-build next from The closets commit to then and it still does not boot. 

This make me think the problem is in one of the dependent repositoroes downloaded by the script? U-Boot perhaps? Not sure how to force earlier versions of those to test.

broadblues, 2025-05-16:
Did some experiments with the u-boot from 1.04 orangepi provided generic images, unfortunately not successful.

I built the latest 1.06 and then installed that on the working image. That stopped it from booting. 

Refering to the platform_install.sh script I wrote the two files from the original image back into the broken image, and restored boot. 

So I then applied those 1.04 files to the custom 1.06 image, unfortunately that still didn't boot.

last step was to apply the files from the 1.06 image in case they hadn't been written correctly by the build script, but again no boot.